parallaxinc/Simple-Libraries #16: unknown type name 'jm_time'

DavidZemon, 2016-01-05:
Where is the type `jm_time` defined? I can't find it with grep and PropWare is failing to compile `Learn/Simple Libraries/Social/libbadgetools/jm_time.c`

AndyLindsay, 2016-01-05:
I don't know.  I removed it from the libbadgetools library, so when compiling with SimpleIDE, it gets skipped.  What happens if you delete it?

AndyLindsay, 2016-01-05:
Sorry for the inconvenience, I'll fully remove jm_time and update tomorrow.  There'll be a few other updates and corrections, but no other large additions lIke today.

DavidZemon, 2016-01-06:
Okay, works for me

AndyLindsay, 2016-01-06:
Alright, I removed jm_time.c, and SimpleIDE didn't seem to notice it was gone when I rebuilt.  Please retry with PropWare.

DavidZemon, 2016-01-07:
jm_time.c still appears in the repo. Are you sure you removed it?

PropWare's blacklist now contains four files: `jm_time.c`, `date_time_epoch.spin`, and `colorpal.c`/`colorpal.h`. This means PropWare does not compile those four files into the library and it doesn't include the header files in the installation.

The spin file I had to ignore because PropWare tries to compile Spin files via spin2cpp's dat extraction feature. It fails on `date_time_epoch.spin` because it doesn't have a DAT section in it. This may or may not be a mistake on your part, so I'm perfectly fine with you leaving that in and PropWare continuing to blacklist it. Someday PropWare will support converting spin -> C++ and then it'll be great to have this Spin file available.

`colorpal.c` contains a compilation error when compiled with XMM. Looks like work was started on XMM support but never finished (I see some XMM-related lines commented out rather than selected via preprocessor macros. See lines 113, 118, and 125.

AndyLindsay, 2016-01-07:
Thanks for the update.  Hmm, that's mysterious; I'm not sure how I let jm_time.c back into the repo.  I just removed jm_time.c, date_time_epoch.spin, and commented the one XMM ifdef I could find in colorpal.c.  The colorpal library doesn't need to be XMM.  Please let me know if everything makes it through this time.  Thanks!

DavidZemon, 2016-01-07:
That did it! Thanks!

The header didn't contain any bad code, but I excluded it as well because I foresaw some poor PropWare user being dumbfounded that colorpal didn't work even though they had the `#include "colorpal.h"` in there and had the exact same code as someone in SimpleIDE

closed 2016-01-07 by DavidZemon