parallaxinc/Simple-Libraries #27: TOGGLE command bug in SimpleTools.h

kgracey, 2016-06-28:
I created the following code from Blockly and had to add the low(26) and low(27) to make it work properly:

```
#include "simpletools.h"

// Describe this function...
void blink26() {
  low(26);
  while(1) {
    toggle(26);
    pause(100);
  }
}

// Describe this function...
void blink27() {
  low(27);
  while(1) {
    toggle(27);
    pause(105);
  }
}

int main() {
  cog_run(blink26, 128);
  cog_run(blink27, 128);
}
```
It seems that SimpleTools.h should default the pin to low so this isn't needed in the code. @AndyLindsay is aware of the issue and proper solution. Until this is fixed I will work around it by having Blockly add the low into the two cogs. - KG

DavidZemon, 2016-08-22:
Looks like this is just a matter of setting the `DIRA` register with `DIRA |= mask;` as the last line in `toggle()`, is that all that is being asked? I'll submit a merge request for this tonight. It seems like a good idea so that `toggle()` is consistent with `high()` and `low()`.

AndyLindsay, 2016-08-23:
Done, and now compiled and ready to test.  Thanks DavidZemon!

closed 2016-08-23 by AndyLindsay