parallaxinc/Simple-Libraries #29: Please add to ColorPal library

MatzElectronics, 2016-08-01:
To make the ColorPal sensor more useful when combined with modules, displays, or sensors that also use color in some way, this function is an attempt to output a "standardized" color value:

`// ColorPal RRGGBB converter

/*
 * Takes the 12-bit red, green, and blue outputs from a ColorPal
 * sensor and generates an approximate RRGGBB (24-bit) output
 * compatible with HTML color codes and some RGB leds.
 */

unsigned int colorPalRRGGBB( int __r, int __g, int __b )
{

  __r = 21 * __r / 11 - 65 - ( __r * __r / 415  );
  __g = 48 * __g / 17 - 70 - ( __g * __g / 170  );
  __b = 8  * __b / 9  - 30 - ( __b * __b / 1680 );
    
  if( __r < 0 ) __r = 0; if( __r > 255 ) r = 255;
  if( __g < 0 ) __g = 0; if( __g > 255 ) g = 255;
  if( __b < 0 ) __b = 0; if( __b > 255 ) b = 255;

  return ((r & 0xFF) << 16 | (b & 0xFF) << 8 | (b & 0xFF));
}
`

AndyLindsay, 2016-08-23:
Done.

zfi, 2016-08-23:
I need a copy of this in the BlocklyProp branch please.

PropGit, 2016-08-25:
Done.

MatzElectronics, 2016-08-29:
@PropGit @zfi Found a typo in this library, submitted PR - @AndyLindsay will merge

AndyLindsay, 2016-08-30:
New commits merged into master.  Thanks Matt!

Andy

closed 2016-08-30 by AndyLindsay

PropGit, 2016-08-30:
@MatzElectronics @AndyLindsay - Was this done through the web interface?  The space character in "Simple Libraries" was translated into %20, resulting in a whole new folder path instead of overwriting the files in the libcolorpal cmm/lmm folders.

[image]

I fixed this and pushed back.

MatzElectronics, 2016-08-30:
It was the web, thanks for the catch!