parallaxinc/Simple-Libraries #39: simpletext writeStr does not appear to send terminating char

MatzElectronics, 2016-09-04:
When I used writeStr() and readStr() together, I had to send a send a 0 (null) character after the string.  Otherwise readStr() filled it's limit with the send string in a sort of loop.

AndyLindsay, 2016-09-04:
A terminating character should be appended to the string that gets sent, typically \r or \n.  This keeps writeStr compatible with more receiving devices.  By automatically adding a 0 to the string that gets sent, it would interfere with devices that expect \n or \r but for which \0x0 has a different meaning.

AndyLindsay, 2016-09-04:
P.S. Off the top of my head, examples of receiving devices where 0x00 has another meaning include the Parallax Serial LCD, XBee, and Parallax software terminals.

MatzElectronics, 2016-09-04:
Good to know.  It's easy to just append the zero.

MatzElectronics, 2016-09-04:
Go ahead and close the issue...  I know now how I want to handle this.
Thanks!

AndyLindsay, 2016-09-04:
Okay, thanks Matt.

closed 2016-09-04 by AndyLindsay