parallaxinc/Simple-Libraries #85: SimpleTools pause() function

JonMcPhalen, 2017-06-02:
I found that the pause() function can be tripped up with large values. May I suggest that it be re-coded as follows?
```
void pause(int time)                          // pause function definition
{ // If st_pauseTicks not initialized, set it up to 1 ms.
  // if(!st_pauseTicks) set_pause_dt(CLKFREQ/1000);
  
  long t0 = CNT;
  
  while (time--) {
    waitcnt(t0 += st_pauseTicks);
  }  
}
```

AndyLindsay, 2017-06-03:
Hmmm, let's see...  That would increase the max pause from 26, 843 ms to almost 24.9 days.  Sounds good to me; thanks Jon!

JonMcPhalen, 2017-06-03:
My pleasure, Andy. It's been a long-time coming, but I am finally forcing myself to work with C on the Propeller. I just finished a commercial app that is 6K lines (Spin); I'm wanting to see if I can do it in CMM and get more speed.

AndyLindsay, 2017-06-03:
Repository updated.  This change will appear in the next distribution.

closed 2017-06-03 by AndyLindsay

AndyLindsay, 2017-06-03:
Cool, sounds like a pretty epic project.  There's a nice Spin to C++ (or C)
converter available called spin2cpp.  It might provide a first-cut on
execution speed with less effort than a by-hand port.

https://github.com/totalspectrum/spin2cpp/releases

Andy

JonMcPhalen, 2017-06-03:
It was tough -- a continually moving target (kind of like the P2!). The client finally stopped making changes, I finished the code, and testing -- so far -- has shown that it's working as desired.

I have that and have played with the Spin-to-C converter, but wouldn't use it on a real project as that would deny me the opportunity to learn and optimize my own libraries.