parallaxinc/Simple-Libraries #169: buttons.c (demo branch)

VonSzarvas, 2018-04-02:
Buttons is broken because some pins are hard-defined.

Suggested update (with original code in comments after each changed line):

 `low(NAV_TOUCH_R); // low(2);
  
  high(NAV_TOUCH_L); // high(4);
  waitcnt(CLKFREQ/4000 + CNT);
  tDecay[1] = rc_time(NAV_TOUCH_L,1); // tDecay[1] = rc_time(4,1);

  high(NAV_TOUCH_R); // high(27);
  waitcnt(CLKFREQ/4000 + CNT);
  tDecay[0] = rc_time(NAV_TOUCH_R,1); // tDecay[0] = rc_time(27,1);`

PropGit, 2018-04-02:
@VonSzarvas - Yes, rc_time should be leaving the pins as input when it's done.  @AndyLindsay can correct me if I'm wrong.

If we later set the pins low, they'll be set back to outputs.  Is that what you intend, so there's no floating button lines (is that the "interference" you mean)?

Question: What's on I/O 2 (that was set low twice in the old code but not at all in the new)?

AndyLindsay, 2018-04-02:
I don't think I wrote that.  The button code for the non-WX badge was this:

`int button( int pad )
{
  char *addr = (char *) tself->p_pinslist;
  int pin = addr[6-pad];
  int ctr = 0;
  for(int i = 0; i < 5; i++)
  {
    high(pin);
    pause(1);
    input(pin);
    pause(5);
    int pb = input(pin);
    if(pb) ctr++;
    low(pin);
    pause(1);
  }
  return !(ctr == 5);
}

int buttons(void)
{
  int pb = 0;
  int mask = tself->pinsmask;
  for(int i = 0; i < 5; i++)
  {
    //high(pin);
    OUTA |= mask;
    DIRA |= mask;
    pause(1);
    //input(pin);
    DIRA &= (~mask);
    pause(5);
    //int pb = input(pin);
    pb |= ~INA;
    //low(pin);
    OUTA &= (~mask);
    DIRA |= mask;
    pause(1);
  }
  char *addr = (char *) tself->p_pinslist;
  int pads = 0, pin;
  for(int i = 0; i < 7; i++)
  {
    pads <<= 1; 
    pin = addr[i];
    int bit = (pb >> pin) & 1;
    pads |= bit;
  }    
  return pads;
}
`

VonSzarvas, 2018-04-02:
Io2 was reassigned to another sensor. Instead the other side of the passive buttons is permanently low.
That links to why I wondered if setting the 2 passive button lines low when not being scanned would remove potential floating problems from the high impedance. Its 470k on the line to gnd.

MatzElectronics, 2018-05-31:
Should be fixed now.  This can be closed.