parallaxinc/Simple-Libraries #173: wifi_print(TCP...,string) only works with string allocated to wifi_buffer

AndyLindsay, 2018-07-06:
If you try to pass a different string, it is not populated with the response.

AndyLindsay, 2018-07-06:
Partially addressed with documentation and example code updates in [v1.4.88 Update wifi_scan(TCP... docs](https://github.com/parallaxinc/Simple-Libraries/commit/4a1bf0e636fb08b062f044df6f2e811f39a7d66b)

AndyLindsay, 2018-07-06:
Check how common content-length is in headers.  If common, then add function to deliver only content.

AndyLindsay, 2018-07-06:
Hmmm, this actually looks pretty doable...

zfi, 2018-07-06:
A word of caution regarding character sets as it relates to the content-length header. If the character set is multi-byte, such as UTF-8, the value in the content-length will be under reported.

Details can be found [here](https://sankhs.com/2016/03/17/content-length-http-headers/)  and [here](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Length).

AndyLindsay, 2018-07-06:
Got it.  Thank you!

AndyLindsay, 2018-07-10:
This should help: int wifi_getNamedValue(char *strSource, char *strName, char *fmt, ...);

Usage example in Dev branch's [Temperature from OpenWeatherMap](https://github.com/parallaxinc/Simple-Libraries/blob/6c2ac9a67ef9c48700c5061dbdb99927ef3475e4/Learn/Examples/Network/WiFi/Temperature%20from%20OpenWeatherMap.c).   
```
    float temp = 0;
    wifi_getNamedValue(str, "temp", "%f", &temp);
    print("temp = %6.2f deg K\r", temp);
```