parallaxinc/Simple-Libraries #176: remainder reserved word

Steph-Parallax, 2018-09-18:
Naming a variable "remainder" throws this compiler error

[image]

[image]

Steph-Parallax, 2018-09-18:
Generated C.

/* SERIAL_TERMINAL USED */

// ------ Libraries and Definitions ------
#include "simpletools.h"

// ------ Global Variables and Objects ------
int quotient;
int remainder;

// ------ Main Program ------
int main() {

  quotient = (10 / 3);
  remainder = (10 % 3);
  print("%s%d%s%d\r", "quotient = ", quotient, " and remainder = ", remainder);

}

Steph-Parallax, 2018-09-18:
Oh, let me try this in Demo. It may actually be caused by a different bug in the print multiple block that isn't in production yet.

Steph-Parallax, 2018-09-18:
Tested in Demo, it gets the same error message. So it is not related to blocklyprop #1538

AndyLindsay, 2018-09-18:
This is in Propeller GCC
#define remainder(x,y)   __TGMATH_REAL_2(x, y, remainder)

I'd recommend adding 'remainder' to the reserved words list in blocklyprop.

Steph-Parallax, 2018-09-18:
Thanks! I'll open an issue over there and link to this.

closed 2018-09-18 by Steph-Parallax