parallaxinc/Simple-Libraries #182: Update gps parser for new NMEA protocol

VonSzarvas, 2019-02-20:
update required in : gps_run.c

GPS strings used to be in the format: $GPxxx  (such as GPRMC)

But now they are: $Gxxxx  (such as GPRMC, GNRMC, GARMC, GLRMC)

Ie. the second character defines which satellite system the data is coming from. It's seamless for the user, but the current code cannot parse strings which do not start $GPxxx. 

// The first 2 chars after the $ sign in the NMEA message represent the Talker ID, representing the source of information:
// GP: GPS; GL: Glonass; GA: Gallileo; GN: combinded GNSS systems. 

Working modified file attached.

Refer to the new byte var : gpsType  (which in retrospect would probably be better as "gpsSystem")

Could the gpsSystem also be made available as a parameter in gps_data, so users can query & display the currently active gps system when getting lat/lon/etc.. data ?

Note: I've not searched the entire GPS code base to see if parsing of NMEA strings is done anywhere else.... This is what I needed for a GPS module demo today, is all.

VonSzarvas, 2019-04-04:
> This is a library issue, and I'll get it resolved soon. It does lend to a possible enhancement for a retrievable value that lets the user know the source (GPS, GLONASS, Galileo) of the incoming data.
> […](#)

@MatzElectronics Any update on this task?

MatzElectronics, 2019-04-09:
@VonSzarvas let us know how testing goes, and if/when we can close this one :)

VonSzarvas, 2019-04-10:
Tested and approved for merge to main.

Summary:

All standard block functions tested ok : Date,Time,Lat,Lon,Alti,Heading,Speed,SatCount

Advanced function (gps_sources) for SatType tested ok  : correctly getting "N" for Glonass/GPS combo in EU zone. Matt reported getting "P" from the US, which implies the new code for SatType parsing is reporting correctly.

VonSzarvas, 2019-05-14:
I believe this issue can be closed / marked as resolved.

AndyLindsay, 2019-05-14:
Thanks Michael.  Closing now...

closed 2019-05-14 by AndyLindsay