parallaxinc/Simple-Libraries #193: Get WiFi IP Address has error

VonSzarvas, 2019-05-14:
Image shows test code. Last line causes the error.

Compile... Failed!

-------- compiler messages --------
Included libraries: badgewxtools, wifi, simpletools
single.c: In function 'wifi_ip_string':
single.c:27:7: error: expected declaration or statement at end of input
single.c:27:7: error: expected declaration or statement at end of input

[image]

Generated code
`// ------ Libraries and Definitions ------
#include "simpletools.h"
#include "badgewxtools.h"
#include "wifi.h"

// ------ Function Declarations ------
char *wifi_ip_string(int __mode);

// ------ Main Program ------
int main() {
  badge_setup();
  wifi_start(31, 30, 115200, WX_ALL_COM);
  // Allow the Badge WX to be programmed over WiFi
  high(17);
  wifi_mode(STA_AP);
  oledprint("Hello4");
  oledprint(wifi_ip_string(STA));

}

// ------ Functions ------
char *wifi_ip_string(int __mode) {
    int __ip[4];
    char __result = wifi_ip(__mode, __ip);
    char *ipStr;
    if (__result == 'E') {
      strcpy(ipStr, "Error          "); // Save string into variable ipStr.} else {sprint(ipStr, "%d.%d.%d.%d", __ip[0], __ip[1], __ip[2], __ip[3]);} return ipStr;}`

PropGit, 2019-05-14:
@VonSzarvas - Are the last two lines really formatted as shown?  They are missing line endings, thus the rest of the block, beginning with '}' is _commented out._

VonSzarvas, 2019-05-14:
[image]

PropGit, 2019-05-14:
Thanks!

This is probably a code generator issue inside BlocklyProp, rather than a Simple Libraries issue.  Let's have @MatzElectronics weigh in and we'll move this issue if necessary.

AndyLindsay, 2019-05-14:
Also, ipStr should point to declared memory that stays in scope while
accessed.  For example, declare a large enough array in main and pass
pointer to wifi_ip_string.

PropGit, 2019-05-14:
Argh!  I missed that <kicking self>.  Thanks @AndyLindsay

AndyLindsay, 2019-05-14:
Sure thing.  That is also something that should be changed in the BlocklyProp code generator, and I agree that this should be moved to [BlocklyProp issues](https://github.com/parallaxinc/BlocklyProp/issues).  Also, @VonSzarvas, here is a link to working [station IP address](https://github.com/parallaxinc/Simple-Libraries/blob/demo/Learn/Examples/Network/WiFi/Check%20Station%20IP%20Address.c) code in SimpleIDE's [examples folder](https://github.com/parallaxinc/Simple-Libraries/tree/demo/Learn/Examples/Network/WiFi).  I tested it yesterday.

PropGit, 2019-05-14:
Transferring to BlocklyProp repo issues.