parallaxinc/Simple-Libraries #198: recv.c

VonSzarvas, 2019-07-03:
wifi_recv command can "block forever" if tcp rx buffer empty when bytes requested.

one issue at a glance.... line 70 looks like a problem.

`if(bytesReady == 0) return 0;`

because in that scenario, line 79 doesn't get called ! 

`wifi_simpletermResume();`

so the debug term will appear to crash.

There may be more to this, and the same issue applies to wifi_scan.
I don't think wifi_recv should block under any circumstances. However- the "block forever" concern may just be an issue for Badge users, as simpleterm is required for all comms.

AndyLindsay, 2019-07-03:
Nice catch.  Do you want to try the fix in demo.blockly.parallax.com?

VonSzarvas, 2019-07-03:
This is the fix thus far. Perhaps wait on demo until it's tested, as may be some other gremlin to satisfy!
Will know in a couple days.

[recv.c.txt](https://github.com/parallaxinc/Simple-Libraries/files/3356412/recv.c.txt)

AndyLindsay, 2019-07-03:
Okay, thanks.  I saw in some other correspondence that sscan might be up to
some mischief too.  Will wait for more news before updating.

AndyLindsay, 2019-07-03:
On second thought, I'll update the Dev branch with anything that needs testing.  You can get the zip with everything else current here:
   https://github.com/parallaxinc/Simple-Libraries/archive/Dev.zip
[Here is info on the latest commit](https://github.com/parallaxinc/Simple-Libraries/commit/35bac9b61bb240fa5f9e0cac4628ec7c2cf771bd)