parallaxinc/Simple-Libraries #200: Solo S3 projects fail to compile - Invalid byte code

zfi, 2019-08-30:
An issue was reported in parallaxinc/solo#74 that has been tracked back to an error in the Simple Libraries. The sample program compiles successfully when version 1.4.122 of the library is used. However, when version 1.4.131 is used, the cloud compiler fails while parsing a file in this library. It is expecting only ASCII characters in the file but is finding a 0xe2 in the file.

Further details available in the Solo issue report.

zfi, 2019-08-30:
The error is coming from:

```
2019-08-30 02:35:49,463 INFO flask.app uWSGIWorker3Core0:Parsing '/opt/parallax/simple-libraries/Learn/Simple Libraries/Robotics/S3/libs3/s3.h' [in ./PropCCompiler.py:212]
```

zfi, 2019-08-30:
After more research, the error is caused by a difference in how Python3 processes text files. This [link](http://python-notes.curiousefficiency.org/en/latest/python3/text_file_processing.html) does a much better job of explaining it. When opening text files, Python is now quite strict in what it expects to find unless it is given a character set parameter. The header file container horizontal line characters, which are, strictly speaking, not text characters.

The cloud compiler has been updated (v1.3.4) to accept the full Latin-1 character set.

MatzElectronics, 2019-11-18:
Should be able to close this now.